cherryphilip74/iSpindel-PCB #2: Tilt not going lower than 0°P

BulletMag, 2019-05-31:
Use a v4 board but had to attach a 10g weight above the gyro board to get the unit to around 27° in plain water, 
I configured using the instructions laid out in the original document and everything was working fine so popped it in a wine I was starting, for the first 48 hours the SG curve was moving as expected until it reached 1.000 (27.06° tilt)and then it remained that way for the last 36 hours.
i read somewhere else (it escapes me where) that you said weights were not needed but i did need to add some to get the unit to the required 26° (ish) tilt 

as there are no instructions for the V4 boards and the gyro is moved to the bottom is the centre of gravity now not being registered correctly?
Previous designs have had the gyro at the pivot point (where the iSpindle breaks the surface of the liquid) but this is much lower.

I was wondering if anyone else has met with this issue?

cherryphilip74, 2019-06-01:
Hi @BulletMag, could you tell me what battery brand did you use?
The 4.0 pcb can it reach the correct tilt if you use a battery with equivalent weight between from 44 to 48gr. Could you share some pic of your iSpindel? I confirm that on this version ballast is not needed and I reached always a tilt around to 25 degree.

BulletMag, 2019-06-01:
I originally brought this as a kit from eBay but after seeing your PCB i decided to have some made up and use that instead, i needed to buy other components such as the 6 pin lipo charger board and the correct battery holder to comply with your BOM, the battery that was supplied to me was branded "UltraFire" 

[image]

I didn't weigh it before i configured and calibrated the unit.
the iSpindle in question is currently in a brew with a couple of days to go so i can't provide photos at this time.

What weight are the batteries you use stand alone ? this way i can shop for a better candidate.

I thank you for your reply and will keep you updated

cherryphilip74, 2019-06-01:
Hi, now is clear for me! Did you use a wrong battery for this pcb, in fact they weight around 22/25gr. You have add a 10 gr ballast for reach the right tilt. Now, you can buy a Samsung 30Q 18650 battery, it weight is 46gr and you can also remove the ballast.

closed 2019-06-01 by cherryphilip74