cherryphilip74/iSpindel-PCB #3: V4 PCB Deep Sleep not waking up

GarySwanepoel, 2019-08-21:
Please can you assist me. I purchased PCB rev 4 Boards from PCBWay. 
My Ispindel sends data to ubidots the first time and then wont wake up again to send data again. 
I have read up comments regarding the Bat 43 diode vs 330 resistor and have tried both with no effect. I am a novice when it comes to building an IOT device and would appreciate it if you could spot any mistakes that I might have made. I have attached some photos and setup. 
This is the second one that I have built with the same problem.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Thanking you in advance

trafim-uakraine, 2019-11-19:
I have already written many times, he will not wake up.  Doit.am should not be written on the module

cherryphilip74, 2019-11-29:
did you solve or not?

trafim-uakraine, 2019-11-29:
No