cherryphilip74/iSpindel-PCB #6: DS18B20 solders damages battery

Naesstrom, 2019-10-23:
I suddenly had some problems with the temp sensor stops working and I think I've found the reason.
On the V4 board the DS18b20 sensor is positioned at the bottom and if the soldering can damage the protective film on batteries. I'm guessing that short circuited my sensor, hoping it will be enoough to just replace it.
[image]
[image]

cswelin, 2019-10-23:
The instructions indicate you should try to keep these flat as possible.

Naesstrom, 2019-10-23:
> The instructions indicate you should try to keep these flat as possible.

I never found and written instructions for the v4 so I used Phillipo's video as a guide, my misstake then! Will try and replace the sensor and hope the rest works!

ErikdBr, 2021-01-18:
Stick the legs of the sensor through the holes and keep the sensor in place by bending the outer two legs. Then cut of the middle one as close to the hole as possible, then solder the middle one, repeat for the other legs. That way you get nice flat soldered dots.

jiblits, 2021-06-01:
Yeah I'm also having this issue. It was blowing the leads on the board and causing the probes to stop working.

I'm thinking wondering if it's possible to do a slight redesign where the battery is moved up slightly. When I get around to fixing my busted boards I'm probably just going to throw some hot glue to provide some extra protection from this kinda damage.

cherryphilip74, 2021-06-02:
The DS18B20 probe sensor was moved outside holderbattery perimeter , just to avoid possible short with wall of the battery.  This improvement is available on new pcb 4.1 (already released).BRcherryphilipInviato dal mio Galaxy
-------- Messaggio originale --------Da: jiblits ***@***.***> Data: 02/06/21  01:39  (GMT+01:00) A: cherryphilip74/iSpindel-PCB ***@***.***> Cc: Subscribed ***@***.***> Oggetto: Re: [cherryphilip74/iSpindel-PCB] DS18B20 solders damages battery (#6) 
Yeah I'm also having this issue. It was blowing the leads on the board and causing the probes to stop working.
I'm thinking wondering if it's possible to do a slight redesign where the battery is moved up slightly. When I get around to fixing my busted boards I'm probably just going to throw some hot glue to provide some extra protection from this kinda damage.