cherryphilip74/iSpindel-PCB #12: Cant find the Gerber Files anymore!

ps915, 2019-12-07:
Where is the gerber file? I just saw you deleted it!

cherryphilip74, 2019-12-07:
I will update with the new version

ps915, 2019-12-07:
I don't want to rush you but i am about to order PCBs from china and would like to add it to my order! =)

When will you have finished the new version?

cherryphilip74, 2019-12-12:
end for year.

ps915, 2019-12-12:
If you are located in europe (i am from germany), i can send you a PCB for testing purposes when i get them from China. I'll be ordering via DHL Express which takes just a few days to arrive! =) 

So if i can support you in this way, i am happy to do so! =)

cswelin, 2019-12-12:
> end for year.

Just of curiosity, what sort of changes have been made?

cherryphilip74, 2019-12-13:
yes, I live in Italy.

Filippo

*Sent:* Thursday, December 12, 2019 at 1:20 PM
*From:* "Phillipp Seidel" <notifications@github.com>
*To:* cherryphilip74/iSpindel-PCB <iSpindel-PCB@noreply.github.com>
*Cc:* cherryphilip <cherryphilip@email.com>, Comment <
comment@noreply.github.com>
*Subject:* Re: [cherryphilip74/iSpindel-PCB] Cant find the Gerber Files
anymore! (#12)

If you are located in europe (i am from germany), i can send you a PCB for
testing purposes when i get them from China. I'll be ordering via DHL
Express which takes just a few days to arrive! =)

So if i can support you in this way, i am happy to do so! =)

ps915, 2019-12-13:
Awesome, so when you have the Gerber Files ready, send it to blog[at]seidel-philipp.de together with your address and i take care of the rest :)

Greetings,
Phil

cswelin, 2019-12-14:
A side note for anyone, this is git so you can go back and pull a previous commit.

cherryphilip74, 2019-12-14:
Hi ,
here the new iSpindel-PCB with GY-521 module mounted directly on pcb. You can order the new version on JLCPCB.com and require smt assembly . I will put the gerber files, BOM and PicknPlace instruction on my github.
[image]

ps915, 2019-12-15:
nice! Do you also have a shematic by any chance! Still can't find the gerber files/bom/pick&place

cswelin, 2019-12-15:
> nice! Do you also have a shematic by any chance! Still can't find the gerber files/bom/pick&place

Go back to https://github.com/cherryphilip74/iSpindel-PCB/tree/iSpindel---SMD, its on another branch.