cherryphilip74/iSpindel-PCB #13: Logging, but unable to reset to re-calibrate

micko247, 2020-01-02:
Hi, I have recently built two units (v3.2) & both are logging. (I actually had a number of issues, have built 10, especially with the temp logging, but I’ll leave that for another post).
I believe I got something wrong with the calibration as the SG stays at 1.004 regardless of the tilt. I use brewfather to log via http. I had no problems accessing configuration until now after they’ve started logging. I cannot access the config even though the wemos does rest and try to open an access point but it logs out. 

Thanks in advance, Mick

cswelin, 2020-01-02:
Append [SG] to the name, since it’s reporting in SG BF requires [SG] in the name.

micko247, 2020-01-02:
Thanks cswelin, do you mean the name of the spindle? Mine are Red and Blue. Do you have a link that describes this? Thanks

cswelin, 2020-01-02:
@micko247 

Correct, in the ispindel configuration, so you’d namw yours Red[SG] and Blue[SG]. I’ve never had luck putting it as a prefix (would still report 1.004), only suffix.  Here’s the doc explanation it https://docs.brewfather.app/integrations/ispindel

micko247, 2020-01-02:
Great thanks. Unfortunately I can’t access configuration. I may try re-loading the firmware. Thanks

cswelin, 2020-01-02:
You should be able to reset to config mode by shorting the two holes on the top of the PCB a couple times In row. (on the part that sticks out)

micko247, 2020-01-02:
I don’t think the v3.2 has this. The reset on the wemos does work, the light starts blinking, and it establishes an access point I can connect to on my computer. But it doesn’t connect successfully. I’ve tried several times on both units.

cswelin, 2020-01-02:
Right, 
Does it connect to the access point but not load the captive portal? if so, try going to `192.168.4.1` in the browser

micko247, 2020-01-02:
Correct, and I’ve tried 192.168.4.1

micko247, 2020-01-02:
Ahhh, now I see the bit about Plato & SG. I’d missed that bit. Cheers

micko247, 2020-01-06:
Specific issue fixed, for some reason Chrome went offline.

closed 2020-01-06 by micko247