cherryphilip74/iSpindel-PCB #14: Using 50ml centrifuge tubes?

fodi666, 2020-01-03:
Hi,
seeing the SMD version I had some hopes that the iSpindel could be redesigned to fit in a [50ml Falcon tube](https://images-na.ssl-images-amazon.com/images/I/31Zpw7D9RQL._SX342_.jpg). I made some measurements:
total external length: 114mm
total length before the conical end, which has full width: 99mm
internal diameter: 27mm

Is it possible to squeeze everything in? Could you prepare a smaller version? I am willing to use a shorter battery if needed (like 14500 which is the size of AA batteries)

Further images of my measurements: [https://imgur.com/a/oD1Yfmx](https://imgur.com/a/oD1Yfmx)

Thanks

cherryphilip74, 2020-01-03:
Yes, i can try..

I need few weeks, at the moment I'm working filling controller.

fodi666, 2020-01-13:
hi,
finally I tried designing one that seems to be working: https://easyeda.com/editor#id=|7bb079167e2c425f9a1a1b8b64ef0805|5c77f5e7a6c54715b207900c2b6ad742|88466c52a5cc43fba402472e6034949a|83aeb81fb016464bba86769db916da22
I removed the charger board to save some space and recreated the GY521 from its components to avoid daughter boards
I also used a bare ESP-12S board instead of the Wemos

cherryphilip74, 2020-01-21:
Link not work

gnalbandian, 2020-02-12:
Did this worked for you? Can you expand on your experience? Thanks!