cherryphilip74/iSpindel-PCB #17: Reply to your last email

jhwerner1, 2020-01-15:
**[Your Quote]()** "After flashing pls you solder the Wemos on pcb... i don't know what is the
issue."

**The issue is...**
After re-installing the Wemos, I still cannot get it to set an access point, cannot re-flash with NODEMCU, logs said "Finding device", never finds.

Followed your Technical Details/ Components instructions to the letter.

Question: Concerning https://jlcpcb.com, when I ordered, they show  images of SMD board with 
Original part placement and  Corrected Part Placement. 

They rotated the D1 Diode 180 degrees, they also rotated the IC2 & IC2 components 90 degrees on the corrected part placement image. Is this causing my issues? If it is how due I prevent them from making changes?

Joe

cherryphilip74, 2020-01-15:
could you send me the DFM analysis that can you find inside to order
history? I want to check if jlcpcb provided to correction or not. The
preview is not real.

Thanks
Filippo

cherryphilip74, 2020-01-15:
Hi @jhwerner1 ,
here a screenshot of layout
[image]
All devices are in correct position, but i want to understand the real issue.
Could you send me some pics of the pcb? Where are you live? We can organize a pickup for shipment at my attention
Thanks and Regards
Filippo

closed 2020-01-15 by cherryphilip74

jhwerner1, 2020-01-15:
I copied and pasted the images from the DFM Analysis into a Word Document
attached to this email.

[image: image.png]
[image: image.png]

Original parts             Actual Placement
placement

cherryphilip74, 2020-01-15:
The actual placement is correct.

jhwerner1, 2020-01-15:
Is this suggestion from an earlier post correct?

Quote "You might want to try to manually force GPIO0 (pin D3 on the board)
low (this initiates firmware upload mode) after reset of the Wemos.

   - Remove USB
   - Connect D3 to GND
   - Plug-in USB
   - Try upload in firmware.
   - disconnect D3 from GND"