cherryphilip74/iSpindel-PCB #18: I can't flash!

rackam74, 2020-01-15:
Hi I've just received the new SMD version.
WHen i finished to build it. I try to flash it.... but it's impossible. Nocemcu programmer is stucked
[image]
I've tried to flash the wemos before soldering.... I can't attach a video.... but with putty there are  some characters that scroll very quickly.
I've already build some of your old pcb, and I didn't have this problem.

cherryphilip74, 2020-01-15:
Hi rackam74, 
could you ship me the pcb? 
Of course shipping at my expense.
Pls send me an email : cherryphilip@email.com

thanks

rjaakke, 2020-01-15:
You might want to try to manually force GPIO0 (pin D3 on the board) low (this initiates firmware upload mode) after reset of the wemos.

- Remove USB
- Connect D3 to GND
- Plug in USB
- Try upload in firmware.
- disconnect D3 from GND

I have 5 of them coming in tomorrow so hope this can be fixed somehow. Still waiting on other parts coming in from China so I can't test it yet.

@cherryphilip74 correct me if I'm wrong on this suggestion.

rackam74, 2020-01-16:
@rjaakke i've send the pcb to @cherryphilip74, so I will check it! I hope it will find the issue!

cswelin, 2020-01-17:
Interested in this well, had a few boards built, once assembled its no longer flashable also scrolls random chars and blue LED stays on.

jhwerner1, 2020-01-17:
Sounds like the same problem I've been having.

cherryphilip74, 2020-01-18:
workaround
https://github.com/cherryphilip74/iSpindel-PCB/issues/19#issue-551776899