cherryphilip74/iSpindel-PCB #19: PCB design issues & work-arounds

jwlusby, 2020-01-18:
I recently built a batch of these and found a couple of issues and found work-arounds

**#1 The 3.3v LDO (IC1) has it's  !EN pin tied to "IN".**  
This caused the LDO to not turn on. Instead, this should be tied to GND or possibly some other pin on the WEMOS so that it could control turning on/off the MPU6050 IC.

The trace with the red X's in this image show what should be remove and the green rectangle shows where a trace should be added.

[image]
**SYMPTOMS** 
- Inability to program WEMOS while connected to the iSpindel PCB.
- Garbage serial output. Appears as if the BAUD rate is wrong. 
- Stuck blue light on reset
- Failure to enter Config mode

**WORKAROUND** I was able to work around the issue be de-soldering the IC, bending the !EN pin up and soldering a small jumper wire between it and the GND pin next to it.

**#2 Pin 8 (VLOGIC) of IC2 (The MPU-6050) does not have a connection to the 3.3V input of VDD (Pin #13)**

[image]

**SYMPTOMS**
- No data from MPU-6050 accelerometer (Data is always 0)

**WORKAROUND:** Solder a jumper from the VLOGIC side of C6 to the 3.3V side of R1.
[image]

cherryphilip74, 2020-01-18:
Hi @jwlusby,
Thanks, thanks a lot!!!
I'm waiting some boards juststart a failure analysis and possible workaround when I did receive email notification with your post.
LDO = type error.. I latched the ouput with EN in pull-up instead of in pull-down (my error)
VLOGIC: I got the model from another sketch where the PIN is connected to GND with capacitor without 3V3. In fact you're right and as reported on DSH this pin must be connect at 3V3. 
(see proto supplies scheme vs other schematic)
[image]
[image]

There are a lot of GY521 modules doesn't work (0.00°), I wonder if the root cause is the same, VLOGIC fault due to V3V missed.

Another question: could you tell me the tilt value got in flat water?

I will update the schematic, when I will be complete the analysis.

Cheers!
Filippo (cherryphilip)

jhwerner1, 2020-01-18:
Will you let us know when you have uploaded the correct files to JLCPCB.COM, I would rather order the correct version of the PCB, then attempting to make the workaround corrections suggested above.
Joe

cherryphilip74, 2020-01-18:
Yes

jwlusby, 2020-01-19:
Thanks for the quick reply. I'm glad that I could help in some way!

It seems like the issues other have reported could very well be a result of the VLOGIC pin not being connected to the 3.3V input.

> Another question: could you tell me the tilt value got in flat water?

With a freshly-built, non-calibrated iSpindel, and using the latest code from [https://github.com/universam1/iSpindel](url), I'm getting a Tilt value of 45.7°

cherryphilip74, 2020-01-20:
> from https://github.com/universam1/iSpindel <http://url>, I'm getting a

Use a battery with weight >= 48gr.
After calibration the tilt is between 22/26

Filippo

jussipu, 2020-01-24:
Is this issue only on smd board?

cherryphilip74, 2020-01-24:
Yes. Bugs fixed, will be release gerber file soon.

rjaakke, 2020-01-24:
To make it a bit more easy to fix the smd board you can also remove the  !EN pin completely.
I thought I wrecked one board but documentation states !EN can be open or tied to GRD.

To fix issue 2 I soldered a jumper from the VLOGIC side of C6 to the 3.3V side of C2.
The C2 is much bigger in size so it was allot easier to solder.

If possible it would be nice if the C2 would be moved to the left or right in future versions. This would make it easier to plug in the micro USB to flash the wemos. With the current layout you have to lift the wemos a bit to allow plugging in the USB connector.

rtemplen, 2020-01-25:
Is this issue applicable to the 4.0 PCB?
https://www.pcbway.com/project/shareproject/iSpndel_4_0.html

cswelin, 2020-01-25:
> Is this issue applicable to the 4.0 PCB?
> https://www.pcbway.com/project/shareproject/iSpndel_4_0.html

No, that isn't a SMD board.

cmckay22, 2020-03-15:
Just as an update it seems there maybe still a small issue with the new SMD boards (Fixed version). I had some arrive this week and still had issues. I worked with cherryphilip74 yesterday and he found a simple workaround. I added a small solder jumper between the bottom two pins on the left side of the ic chip (pin 2 GND and pin 2 E) and lifted the GND on the right (bottom pin). I accidentally broke the pin on the right side and it still works so If you have soldering skills like me don’t worry about it.

[image]

rjaakke, 2020-04-29:
@cherryphilip74 was this a production fault or does the gerber need an update?