cherryphilip74/iSpindel-PCB #23: No link or file

DrBabbage, 2020-07-22:
Hey there, I dropped you a donation. Would it be possible for you send me the gerber files?
I would also love to have a BOM. 
Nice project!

cherryphilip74, 2020-07-22:
Thanks a lot for donation.Im out of house, when I'll come back I will send you gerber files or you can also order pcb on pcbway.com (shared projects)

DrBabbage, 2020-07-22:
Thank you SO much!
Will order them from jlcpcb so a gerber file would be awesome.
Do you know roughly when you are home again?
Don't stress yourself, I am working at a hospital and thanks to corona I have no time to do anything but working for at least 2 months.

DrBabbage, 2020-08-11:
hey, short question are you at home again?
Thank you so much!

cherryphilip74, 2020-08-11:
No! I'll come back after the Aug 17th.Filippo

DrBabbage, 2020-12-03:
Could you send me the file ? :)
Thanks!